marian42/octoprint-preheat #1: Feature Request: Bed preheat

olehs, 2017-11-21:
Please add a heated bed preheat (commands M140, M190)

olehs, 2017-11-21:
Thanks again. And one more thing...
What about bed cooldown ?

mark-hahn, 2017-11-26:
I haven't tried this yet so I don't know how it works, but I personally would like the bed to just always preheat/cool when the nozzle does.

marian42, 2017-11-26:
I agree. It works like that for preheating. I'm about to implement bed cooldown, currently it only cools tool0.

olehs, 2017-11-26:
@mark-hahn Well, I know what you mean. Sometimes I also need to cool down only nozzle. But there still is separate button for this on OctoPrint's Temperature tab.
I think it would be great if there was a button with combobox instead of Cooldown button in this plugin

mark-hahn, 2017-11-26:
You mentioned tool0.  If my file uses only tool1 will everything work right?

marian42, 2017-11-26:
@mark-hahn Preheating should work in that case. I haven't tested that though, since I have only one nozzle. Cooldown would not work in that case, but I'm working on a fix.

commit b6c94a3d (2017-11-21): Rewrite gcode parsing to handle more keywords, multiple tools and bed heating
references #2
fixes #1
commit 06190551 (2017-11-27): Update cooldown to set temperature for all tools and bed
fixes #1
closed 2017-11-27 by marian42