marian42/octoprint-preheat #4: Doesn't work

reggaemanu, 2017-11-22:
Hi,

I really like the idea of this plugin as I always preheat manually before starting a print but I can't make it works, I've tried with many gcode files (from S3d and Cura) and it always throw me a "Could not find a preheat command in the gcode file." even if there is M104 lines in the gcode files.
What can be wrong?

(I'm using latest octoprint 1.3.5)

marian42, 2017-11-22:
Sorry to hear that. When the plugin first released, there was no update hook, so if you installed on the first day, please remove the plugin and install it again. Once you do that, it should update automatically.

If that doesn't solve it, please send me a gcode file that doesn't work and I'll have a look.

reggaemanu, 2017-11-27:
I've indeed installed it the first day !
I can't mess with it right because I've an ongoing print but I'll test that as soon as it finish and let you know if that works, thanks ;)

reggaemanu, 2017-11-28:
It worked on the printer I have at home (I had the same issue on this one the one at my workshop), can't tell for the one at my workshop as there's still 16h until print end but that should be ok :)
Thanks for your response and a big thanks for this plugin !

PS: not related to this ticket but the cool down button doesn't stop the heatbed

closed 2017-11-28 by reggaemanu

marian42, 2017-11-28:
> PS: not related to this ticket but the cool down button doesn't stop the heatbed

I did fix that yesterday, it will be in the next release 👍

commit 65d0a31c (2018-04-01): Update button at app startup
fixes #4