marian42/octoprint-preheat #8: Messed up State View

n8name, 2017-11-30:
Plugin messes up div state_wrapper. OctoPrint Version 1.3.5
[image]

marian42, 2017-11-30:
Are you using TouchUI?

n8name, 2017-11-30:
Yes, I do. Is there a known conflict between these two plugins?

marian42, 2017-11-30:
No, this is my attempty to make my plugin compatible with TouchUI. Are you using a theme? When I installed TouchUI I got a dark flat design. I added some css rules to work with that design and these rules mess up the view in your screenshot.

n8name, 2017-12-01:
Both OctoPrint and TouchUI themes are the default ones. Can you provide a workaround? 
Honestly I have not used TouchUI that often simply because I use the Telegram Plugin.

marian42, 2017-12-01:
I would like to know how to reproduce your problem so I can fix it. When I installed TouchUI, it looked like the screenshots [here](http://plugins.octoprint.org/plugins/touchui/), not like the screenshot you provided. Is it possible you have TouchUI installed but disabled?

Uninstalling TouchUI would be a workaround.

marian42, 2017-12-01:
Nevermind, I just realized you can have TouchUI installed but still use the default layout. I'll fix this later today.

commit fe24bca7 (2017-12-02): Use css rules to ensure compatibility with TouchUI
fixes #8
closed 2017-12-02 by marian42