marian42/octoprint-preheat #9: "No preheat command in file" on every print from Cura

Stady234, 2017-12-01:
Every time i send new job to octoprint from cura 3.0.4, the plugin button when clicked states that there is no preheat command found.

NeoTrace82, 2017-12-01:
I do have the same issue and wanted to create an issue, ill upload a gcode file from a current CURA 3.0.4 Installation in a second

NeoTrace82, 2017-12-01:
[AA_DMG_v1.1.zip](https://github.com/marian42/octoprint-preheat/files/1520783/AA_DMG_v1.1.zip)

marian42, 2017-12-02:
I just tested the file and it works as expected for me. Can you confirm that the error still appears after removing and reinstalling the plugin?

Stady234, 2017-12-03:
After re-install, everything is working as it should.  Not sure why it acted like that...

NeoTrace82, 2017-12-03:
Same for me did reinstall, now it works

closed 2017-12-03 by marian42