marian42/octoprint-preheat #11: Error updating to 0.20

theDrsh, 2017-12-07:
Hello!

Love the plugin, tried to install latest update and have been getting errors. Tried rebooting and trying again with no luck.

Logs in question:
2017-12-07 09:30:43,358 - octoprint.plugins.softwareupdate - INFO - Starting update of preheat to 0.2.0...
2017-12-07 09:30:43,372 - octoprint.plugins.softwareupdate - ERROR - Update of preheat can not be performed
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 893, in _perform_update
    update_result = updater.perform_update(target, populated_check, target_version, log_cb=self._log, online=online)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/softwareupdate/updaters/pip.py", line 69, in perform_update
    install_arg = check["pip"].format(target_version=target_version)
KeyError: 'target'

marian42, 2017-12-07:
While I'm working on a fix, you can try to uninstall and reinstall the plugin, that should get you the latest version.

theDrsh, 2017-12-07:
That fixed it, Hopefully those logs were helpful.

closed 2019-02-03 by marian42