marian42/octoprint-preheat #12: sequence of heating

Tinchus2009, 2018-01-01:
Hi. Nice plug in but hav a comment: if you want to pre heat for abs for example, bed will have to reach something  between 90 degrees, even 100 degrees. To reach that temperature, you will have to wait like at least 5 minutes.
Right now the pre heat plugin starts both the bed and hotend at the sme time: bed will heat to 100 degrees and will take like 5 minutes at least. Hotend will only take1.5 minutes to reach the 230 dregrees... it is not good to have 4 or 5 minutes your abs cooking inside the noozle till the bed reachs its temperatur
I thing it would be better if the plugin first heats the bed, and once desired temperature is reaches, starts heating the hotend.

marian42, 2018-01-01:
I currently don't have a heated bed but I'll get one soon. Once I have it set up I'll look into this.

Tinchus2009, 2018-01-02:
well. i guess the change is somehow simple, in the plugin code, just change
the gcode used to heat the heated be by the one used to "heat and wait till
is heated"

marian42, 2018-01-02:
Good idea, I hadn't thought of that. However, without testing, I don't know if running two gcode commands that take multiple minutes to execute will work as expected.

Tinchus2009, 2018-01-02:
should work. im not telling the gcode command because i dont rememvet but
the gcide will heat the bed and firmware will wait till that temp
iscreached to execute the code fot hitend (that is how most slucers work
now)

dzenbot, 2018-01-03:
Maybe, front-end wise, the default button would still be "Preheat", and with the use of a dropdown you could offer 2 options:
- "Preheat all"
- "Preheat bed first"

I do think this applies to ABS and PETG and other complicated materials which require hight bed temperatures.

kantlivelong, 2018-01-03:
@marian42 you can use [Virtual Printer](http://docs.octoprint.org/en/maintenance/development/virtual_printer.html) plugin provided by OctoPrint.

allenrowand, 2018-04-07:
Came here looking for the same thing; I'd like to heat the bed first, so that the hotend doesn't drool filament while the bed heats.

The heating gcode Cura writes in my files is:
M190 S60
M104 S205
M109 S205

But the hotend still heats first.

marian42, 2019-02-03:
I know it's been more than a year, but I just added the feature.

closed 2019-02-03 by marian42