marian42/octoprint-preheat #13: Pre-PR Questions

kantlivelong, 2018-01-22:
Hey @marian42,

My recent PR to OctoPrint that adds support for injecting variables into G-Code scripts was accepted and should be in version 1.3.7. I've taken it upon myself to add support tho this plugin but it turned into a re-write.

My re-write allows users to modify the preheat snippet and add it to any other G-Code scripts where they see fit (e.g. beforePrintStarted). The default snippet is designed to allow quick heat-up of the bed(if enabled in OctoPrint) and all tools.

I also re-wrote how gco files are scanned for temp changes by parsing all M(104|109|140|190) and T# (tool changes) until the first extrusion is detected.

Take a look [here](https://github.com/kantlivelong/octoprint-preheat/tree/rework_and_new_features). Please keep in mind that it is not 100% complete but pretty close.

My questions for you are:
1. Are you willing to merge these changes or would it be better that I fork?
2. Should this plugin really support SD at all? In my opinion there isn't much of a gain unless users always use the same temps no matter what but I feel that it is a bad assumption to make.
3. I adjusted it on my end but I was curious if you realized the Print button was hidden behind Pause which prevents prints from running at all. You can even see it in your screenshot. I positioned the preheat/cooldown below cancel but it may be safe to place it over cancel when not printing/paused.

marian42, 2018-01-22:
Hi, thank you for your work!

1. I have no problem discarding my code and merging new code, as long as it retains the current features.

2. I'm not sure, as I don't use SD cards myself. Shouldn't it be possible to just put constant temperatures in the new preheat snippet?

kantlivelong, 2018-01-22:
1. Cool :+1:. If you have time give it a shot and let me know what you think. The current code is functional with the exception of SD fallbacks and the preheat/cooldown button placement

2. Yeah I can certainly retain that setting in a similar manner.

3. I know I did a strike-through on this comment but does the bit about replacing the Cancel button with Preheat/Cooldown when not printing/paused make sense? I can't think of a case where that would cause a problem but always looking for second opinions.

marian42, 2018-01-22:
Replacing the cancel button with the preheat button sounds like a good idea. I liked the large print button but I guess that's not something that should be addressed by this plugin anyway.

kantlivelong, 2018-01-25:
@marian42 

I'm at the point of implementing temperature fallbacks but cannot think of a way to handle the scenario of having multiple hotends. Since we cannot read the file from SD there is no way of knowing which hotened to heat up. Heating both when only one is needed would result in oozing or other issues. Heating one may result in heating the wrong one (e.g. the job is for tool 2 only).

Is there a way that you want this handled? I'm still for not supporting SD at all and simply throwing an error in the UI when temps (are|can) not be detected.

marian42, 2018-01-25:
Since I don't have multiple hotends, I don't know what's best. Just implement any solution you like. Once someone with this scenario voices a preference here, we can still change it. IMO it would also be ok to show an error message and let the user change the gcode snippet to replace the variables with constants.

closed 2019-02-03 by marian42