marian42/octoprint-preheat #18: New API to select file and read temps

gdombiak, 2018-12-19:
I'm the developer of [OctoPod](https://github.com/gdombiak/OctoPod) and got a [request](https://github.com/gdombiak/OctoPod/issues/157) to add support for your cool plugin. I was looking at the code and noticed that it would be convenient if there is a new API endpoint that accepts the path to a file so it gets selected and the temperatures are read for that file. If there is an error doing this, return a proper HTTP status code (besides logging the exception as it does today) so the app can communicate the problem to the end user.

OctoPod would let people browse files, call this new API before calling the existing API that lets you start the preheat.

Thanks,
Gaston

marian42, 2019-02-03:
Why would you want to read temperatures from a file other than to trigger the preheat command? I don't quite understand what the advantage is compared to triggering the preheat command right away.

marian42, 2019-07-26:
Closing this for now, once you provide further details, I'll reopen.

closed 2019-07-26 by marian42

Stady234, 2020-01-21:
I was asked to help give context from a user perspective, as i had asked on the status from the developer of the Octopod;

Basically, from the phone APP or watch APP, it would be nice to allow a file to be "loaded" and show the temps that will be used, and conversely if there was some issue with the gcode... i.e. copied from someone other that the current user, that didnt save the correct temps or the temps are missing completely... then a notification of some sort would notify the end user that this is the situation... or if everything is good, then the preheat button would then be available for remote call through the phone or watch app....

While this has been rare case, t would be nice as a "fallback" check, like you have on the web interface settings of enabling temp if none were given in the file.  but again, to allow a notification that feature is utilized due to file issues.
-
I hope this make sense.