marian42/octoprint-preheat #20: Terminal Command for Preheat

danowar, 2019-01-05:
Would it be possible to add a terminal command that mimics pushing the preheat button?  

This would make it easy to integrate the preheat functionality into other interfaces, such as Printoid.

Stady234, 2019-01-28:
Just wondering but are you talking about the commands sent as soon as you hit the button?  cause this is what mine sent..

Send: M140 S50.0
Recv: ok
Send: M104 S215.0
Recv: ok

marian42, 2019-02-03:
@danowar You can trigger the preheat command from the command line like this:

`octoprint client --httpuser ***** --httppass ***** post_json /api/plugin/preheat "{\"command\":\"preheat\"}"`

[Printoid lets you make custom HTTP requests](https://printoid.net/2017/08/01/trick-20-send-your-own-http-commands-with-printoid-curl-like/).
You need to make a post request to `/api/plugin/preheat` with `{"command": "preheat}` as the payload. I haven't tested it but it should work.
Let me know if you have any questions.

danowar, 2019-02-09:
@Stady234 I realize that is the the commands that are sent, however, those commands would change with each print of different material.

@marian42 Thank you for the tip, that approach does indeed work.  I can now enjoy one-button preheating in Printoid.

closed 2019-02-09 by danowar