marian42/octoprint-preheat #23: [BUG or Feature Request?] Preheat quits before reaches bed temperature

codejedi365, 2019-03-05:
@marian42 First off I love your plugin! It saves so much time!

I just switched over from PLA to PETG printing and I ran into an issue where the Preheat "times out" before reaching the desired temperature.  I'm running my 310x310mm bed up to 85C for PETG and given I'm only running a 12V bed it takes some time to reach the operating temperature.  My bed makes it all the way to about 75C before Preheat expires and shuts off/cools.  

At first thought, I'm thinking this might be a bug since it hadn't yet reached my target bed temperature nor had the extruder heater turned on yet.  I thought I also might of read that Preheat times out after it has been idle at temperature for 10 minutes or so?

Potentially, this might instead be a safety feature you built in which I appreciate the thought but I wonder if you could make it user programmable for situations like mine?  A user defined timeout if it doesn't reach the temperature in the alloted time.

Thanks again!

marian42, 2019-03-05:
Are you using the sequenced preheat feature (it is disabled by default)?

This plugin has no timeout, I'm not sure if Octoprint has a timeout. What does "times out" mean? Does it go back to 0°?

codejedi365, 2019-03-06:
@marian42 yes I am using the sequenced preheat feature... because its awesome LOL.  I'll try it here shortly without the sequenced feature to isolate the issue especially since it doesn't have a timeout built in.

I do not believe Octoprint has a time out either.  It means the target bed temperature goes from "85C" to "off" at a seemingly a specific timeframe after heat initializes. Based on the temperature graph it looks like between 10minutes and 15minutes.  I'll try to time it as well tonight.  Before I wrote this issue up I attempted this a couple times to see if it was recurring and it looked almost identical in the graph.

codejedi365, 2019-03-06:
@marian42 -- so it is exactly the 10 minute mark.  I timed it on my phone twice.  First time I recreated the original problem again just with a timer running.  The second iteration I swapped off the sequencing feature and oddly the same timeout issue happened again at exactly 10 minutes.  I have included 2 snapshots to show how close it was to the desired temperature.

To help rule other errors out, I have re-clicked the preheat button before it cooled all the way down and it made it to the target temperature and stayed on for 5 minutes prior to when I started a print.  Additionally I use Klipper as my printing software/firmware and it has no errors posted to its log so I would rule out temperature max limits or similar error.

[image]
[image]

marian42, 2019-03-06:
What happens if you set the temperature manually in the Temperature tab? Does it turn off after ten minutes? I suspect that this is a "feature" of the printer firmware to prevent fires when you forget to turn off the printer.

codejedi365, 2019-03-08:
@marian42 good catch.  It does turn off after 10 minutes when I use the temperature tab.  I don't understand why it will not idle out if it makes it to the target temperature before 10min.  It is a feature built into my firmware.  I use Klipper and come to find out there was an issue to match and answer this error. https://github.com/KevinOConnor/klipper/issues/695

I almost suggested using the M190 command in your code instead but realized that causes a wait loop at the firmware which prevents any other command like "Cool" to be applied.  I see why you did that and seems to me like M190 is useless/too risky then.  Good work on your part to not rely on that command.

So anyway, I'm closing this issue since it is firmware built in issue with a fix that is a firmware configuration "idle timeout" change to a longer time limit fixes my problem.  Thanks for thinking through this!

closed 2019-03-08 by codejedi365