marian42/octoprint-preheat #24: Add Alert after preheating done

R0dri, 2019-05-22:
Well, I use this every single day. I can't believe it is not a default implementation on the builtin functions of octoprint. 

Anyhow. I find myself using this whenever I what to print something and i'm too impatient to calibrate and then wait for it to heat up to print. Hence, I use the preheat button so that I can control my printer while it is heating. But in the other hand I could waste about 20 minutes after I have done calibrating it and I forget to set to print, even if it has already done heating up.

Could you add an option to send an alarm? Maybe something simple like sending a Gcode to de printer would do.

That could be easily integrated with other plugins like https://plugins.octoprint.org/plugins/M117SpeechSynthesis/ if it could send an M117 after preheating and an M300 if using an LCD with a piezo on it. Or just any gcode for that matter.

Something more advance could be a trigger that could be read by IFTTT or another plugin like https://plugins.octoprint.org/plugins/enclosure/. I think this could be done by just making a simple call to the api of the mentioned above. 

Probably the best way to do this would be letting the user decide some sort of behavior to be expected after preheating is done.. maybe even begin the print job.

marian42, 2019-05-24:
Thank you! I'll look into this once I find the time.

marian42, 2019-05-28:
Do you want the sound to be played in the browser or on the printer or both?
cc @alexisju

marian42, 2019-05-28:
> Probably the best way to do this would be letting the user decide some sort of behavior to be expected after preheating is done.. maybe even begin the print job

Is there any case where it would make sense to use the preheat button for that instead of just starting the print directly?

marian42, 2019-06-04:
@R0dri @alexisju I have implemented a feature where it shows a popup and runs customizable gcode upon preheat completion. Please let me know if this is sufficient so I can push the upddate.

marian42, 2019-07-26:
I just pushed the update, this feature is now live.

closed 2019-07-26 by marian42

R0dri, 2019-09-26:
Wow, been of GitHub for too long. Thanks! Much appreciated!!!