marian42/octoprint-preheat #26: preheat without file loaded

seltix5, 2019-10-13:
hello,

wold it be possible to preheat without loading a file first?

thanks!

marian42, 2019-10-13:
This would be very close to just setting a temperature manually. The point of the plugin is to read the temperatures from the file. But I can add the option to use the fallback temperatures in this case, instead of disabling the button.

seltix5, 2019-10-13:
ohhh i did not know that part! my usage was just to preheat the printer to a base temperature (maybe chosen in the settings) for when I wold send the print file the printer wold just adjust to the print temperature (if not the same).

for example my procedure :
i want to print something but I dont have anything ready. with this option I could just start preheating the printer before or during the slicing process.
Without this option available I wold need to load a random gcode or wait to have my gcode ready but then I wold not preheat, I wold just print...

seltix5, 2019-10-13:
PS : I also wold like to use this for filament change (since you already have a option to popup and beep after preheating) it wold be perfect

tablatronix, 2019-11-02:
Ditto

marian42, 2019-11-03:
I implemented the feature. To use it, you need to enable it in the settings. Let me know if you have any issues.

closed 2019-11-03 by marian42

seltix5, 2019-11-03:
its working great, thanks!