marian42/octoprint-preheat #28: Greyed out, doesnt work.

tablatronix, 2019-11-02:
I don't get it.. Its greyed out, doesn't seem to work.

[image]

marian42, 2019-11-02:
You need to select a file first.

tablatronix, 2019-11-02:
ok, so what is the point of preheat if you have already have a file to print ?

I don't get it.. haha

marian42, 2019-11-02:
I found that I often select a file and manually set the temperature to the printing temperature of the file. While it heats, I do other things like change the filament. When I'm ready and the printer is heated up, I start the print.

This plugin automates that. It reads the printing temperature from the file. That's why you need to select a file, it will set the printer to the temperature that the file will be printed with.

tablatronix, 2019-11-02:
I see, its "Job preheat" , But i saw the defaults in the settings so assumed it could be used as a generic preheat also. i usually preheat when I am still working in fusion or iterating stuff, or slicing in cura, so I do not have a file to print yet.

tablatronix, 2019-11-02:
So is this just a limitation you put in?

can I just mod this?

```PY
if (self._printer.get_current_job()["file"]["path"] == None):
   temperatures = self.get_fallback_temperatures()
```

marian42, 2019-11-02:
I guess this could be added, it was also requested in #26.

marian42, 2019-11-03:
It is now possible to preheat without selecting a file.

closed 2019-12-02 by marian42