marian42/octoprint-preheat #30: Enhancement request: Trigger automatically when file is selected to print

Q-Diddy, 2019-12-06:
Love the plugin and it works great! I am writing to offer an enhancement. In settings it would be nice to have the option to have the plugin automatically run when a file is selected to print. I currently have to wait for the preheat to be done and then hit the print button. If this was done in one shot it would be added convenience.

marian42, 2019-12-07:
I don't completely understand your request. Are you asking for a feature that starts the preheat process when a file is selected or are you asking for a feature that stars the print when preheating is done?

Stady234, 2019-12-07:
> In settings it would be nice to have the option to have the plugin automatically run when a file is selected to print. 

That is a bit counter intuitive to the nature of the plugin.  When you select the print file, the factory way is to just click print to start the heating process... it will start the hotend then the bed (in most slicers)...

> I currently have to wait for the preheat to be done and then hit the print button. If this was done in one shot it would be added convenience.

If you are really not wanting to wait... click preheat plugin after file load... wait 2-5 seconds... then just click print... the heating of both the hotend and bed have already started, so when you click print, it will just continue and start the print like normal..

I have used that "quick" procedure a few times when in a bit of a rush to start a print... however, this really and truly is not utilizing the plugin as intended...

Q-Diddy, 2019-12-07:
@marian42 , sorry for the confusion. Yes, I was asking for the ability to start printing after the preheat has completed. Please see comment below for reasoning behind my ask. 

@Stady234, Thank you for the tip. I will try that. The reason for my ask is that I run bed leveling before every print and I would like to have the bed at the temp I am printing for the leveling process. So heat bed, then start print which runs the beginning script in octoprint that includes bed leveling and purging hotend gcode, and finally start printing. I have different bed temps for different materials and brands of materials.

marian42, 2019-12-07:
Am I understanding correctly that just starting the print would first do the bed leveling and then heat, and you want to heat first and then do the bed leveling instead?

Q-Diddy, 2019-12-08:
@marian42 thats correct. I want to heat the bed first at the temperature set in the gcode file I am printing and then do bed leveling.

Q-Diddy, 2019-12-14:
Closing this request as I realize it's probably not a good idea.

closed 2019-12-14 by Q-Diddy

marian42, 2019-12-14:
Sorry for not responding earlier. From your description, I understand why this feature would be useful and I was planning to implement it. If you change your mind and want this feature to be added, let me know, I think its usecase would be pretty niche but also not a lot of effort to implement.