marian42/octoprint-preheat #31: Preheat button does not allow gcode scripts

brandonscholet, 2019-12-31:
preheat button ignores M80/M81 power supply scripts.

Would love for the plugin to allow start and stop gcode scripts

marian42, 2020-01-01:
What do you mean with "ignores"? What do you expect to happen that doesn't happen?

KriLL3, 2020-01-10:
Ignores is the wrong word, it doesn't support adding g-code on clicking pre-heat, which is a problem if you like me and other guy/gal has our PSUs controlled, if I hit the preheat button atm and forget to also turn on the PSU using something else it doesn't heat up and the printer might error out because it detects that the bed heater is supposedly on but the temps aren't going up. Would be nice to either add a checkbox to send the M80 command to turn on PSU, or better yet include custom g-code field for what happens in addition to setting temps when you click the button.

Stady234, 2020-01-11:
@KriLL3 Why not add the M80 (or any other specific gcode) that you used to turn on the PSU (or other devices like it) to the "Before print job starts" gcode section on Octoprint?

brandonscholet, 2020-01-11:
@Stady234 That's the problem. If you press the preheat button then it does not untilize the give scripts section

marian42, 2020-01-11:
Ok, I can add a feature to perform the gcode script when the preheat button is pressed. I'm quite busy right now, so it might take some time, but it's a good feature suggestion.

brandonscholet, 2020-01-11:
@marian42 sorry, the plug-in does not ultize gcode scripts for when a print starts or ends, or the ability to add actions. I use M80/M81 to turn my power supply on and off. If there was a way to add a gcode section for preheat/cooldown that would be amazing

closed 2020-03-04 by marian42

marian42, 2020-03-04:
I just added an option to run custom gcode when the Preheat button is pressed before the temeperatures are set. Let me know if this solves the problem or if you need something else.

commit 42063c0f (2020-03-04): Add an option to run a gcode script when the Preheat button is pressed

fixes #31
fixes #32