marian42/octoprint-preheat #32: Add fan settings for preheating

ZhenyaKa, 2020-01-19:
When preheating with no fan on fan duct heating very strong and deforming.

Pls add option for fan speed settings when preheating is on!

closed 2020-03-04 by marian42

marian42, 2020-03-04:
I just added an option to run custom gcode when the Preheat button is pressed before the temeperatures are set. You could add a command there to set the fan speed. Let me know if this solves the problem or if you need something else.

ZhenyaKa, 2020-03-04:
> I just added an option to run custom gcode when the Preheat button is pressed before the temeperatures are set. You could add a command there to set the fan speed. Let me know if this solves the problem or if you need something else.

Yes it's great solution.
It fine to add the same options when printing end and power on fan while temp go down to 30-40C for example, but I think your good addon can't solve that problem :(

commit 42063c0f (2020-03-04): Add an option to run a gcode script when the Preheat button is pressed

fixes #31
fixes #32