marian42/octoprint-preheat #34: Preheat Button throwing "Internal Server Error" when trying to use it.

Zinip, 2020-03-09:
Pretty much what the title says, and it's an error I've never seen before. I hadn't installed any new plugins that it could possibly be conflicting with it, but I did recently install Matplotlib and Numpy to finally get Bedlevelvisualizier to function, updated to the 1.4 stable release of Octoprint, and installed Python3 / Pip3. But according to your page it looks like your plugin should work with any Python below 4 if I read things correctly, so I'm at a loss. I'm attatching the error I'm getting, and the log entry for said error below...

This is the error I'm getting:
[image]

And here is the log entry for the error:

```
2020-03-08` 22:12:13,701 - octoprint.plugins.preheat - INFO - Preheating toolx to 215.0.
2020-03-08 22:12:13,703 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin preheat
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 109, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_preheat/__init__.py", line 255, in on_api_command
    self.preheat()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_preheat/__init__.py", line 248, in preheat
    self.preheat_immediately(preheat_temperatures)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_preheat/__init__.py", line 231, in preheat_immediately
    self._printer.set_temperature(tool, target)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/__init__.py", line 474, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/printer/standard.py", line 411, in set_temperature
    raise ValueError("heater must match \"tool[0-9]+\", \"bed\" or \"chamber\": {heater}".format(heater=heater))
ValueError: heater must match "tool[0-9]+", "bed" or "chamber": toolx
2020-03-08 22:12:13,721 - tornado.access - ERROR - 500 POST /api/plugin/preheat (::ffff:192.168.200.60) `61.80ms
```

The "toolx" mentions are interesting. I do have a MMU2S so I have 5 "tools" listed in printer profile, but since it's only one nozzle I have the box checked, and had been using the button with my MMU2S for a long time. However I don't believe anything to be overloaded, because I can use the other ways of preheating my printer.

Also, ty for your wonderful button. It's very much appreciated!

kylix-rd, 2020-03-12:
GCode generated for an MMU2S in Single filament mode will contain the "Tx" command. This tells the printer to prompt the user to select which MMU2S filament to use.

The parse_line() function probably should check that the tool value is actually a valid number. IOW, the "Tx" GCode command should be ignored.

Zinip, 2020-03-12:
Ahh, I know there's a Tx and Tc in every generated single mode MMU2S gcode file (unless you go back in and edit it so you can actually start a print job via Telegram while outside your homes network with no open ports or ddns configured :p ) but it hadn't occurred to me of the possibility of the preheat button's code picking that MMU2S filament selection prompt up as a potential tool and getting confused by it.

But I just tested it and you've nailed it... I sliced and sent a multi-filament print job to Octoprint, loaded it, and the Preheat Button worked! So I did some fiddling around since I knew stock SM MMU gcode's "Start G-Code" generated with PrusaSlicer has  M104 > M140, then the Tx, followed by M109 > M190. Moving the Tx after the "wait for temps" mcodes also allows the Preheat Button to function normally. 

Tysm! It was driving me crazy not knowing why it wouldn't work. I'm going to edit the Start G-Code portion of Custom G-Code in my printer's slicer settings for a work around for the time being... I am going to leave this Issue open though, in hopes of it getting fixed in the plugin's code at some point, and at least now if other people are having this issue they'll know what to do until it's fixed also.

kylix-rd, 2020-03-12:
I hacked the main `__init__.py` file in the plugin with this patch:
```
+++ __init__.py 2020-03-11 21:22:19.095322204 -0700
@@ -70,7 +70,12 @@
                                        self._logger.warn("Error parsing heat command: {}".format(line))
                                        pass
                        if item.startswith("T"):
-                               tool = "tool" + item[1:].strip()
+                               try:
+                                       value = int(item[1:].strip())
+                                       tool = "tool" + str(value)
+                               except ValueError:
+                                       self._logger.warn("Error parsing tool command: {}".format(line))
+                                       pass

                return tool, temperature

@@ -91,7 +96,13 @@
                                        if line == "":
                                                break
                                        if line.startswith("T"): # Select tool
-                                               tool = "tool" + strip_comment(line)[1:].strip()
+                                               try:
+                                                       tool_num = int(strip_commit(line)[1:].strip())
+                                               except ValueError:
+                                                       self._logger.warn("Error parsing tool command: {}".format(line))
+                                                       tool_num = 0
+                                                       pass
+                                               tool = "tool" + str(tool_num)
                                                if tool == "tool":
                                                        tool = "tool0"
                                        if enable_tool and (line.startswith("M104") or line.startswith("M109")): # Set tool temperature
@@ -292,4 +303,4 @@
 __plugin_hooks__ = {
        "octoprint.plugin.softwareupdate.check_config": __plugin_implementation__.get_update_information,
        "octoprint.comm.protocol.scripts": __plugin_implementation__.get_gcode_script_variables
```

Zinip, 2020-03-13:
Yeah you did! Tyssssm! I made the changes you listed and now the preheat button is working flawlessly again. Were you having the same problem with a MMU2S and figured out how to fix the code, or did you just figure it out for my/other MMU2S users sake?

closed 2020-03-13 by Zinip

kylix-rd, 2020-03-14:
Yes, I have an MMU2S and was able to figure this out. Just be warned that if the plugin is updated without this fix (or another, better one), it will break again.

smitty97, 2020-03-14:
patch is failing for me

patching file __init__.py
Hunk #1 FAILED at 70.
Hunk #2 FAILED at 91.
patch: **** malformed patch at line 34:

kylix-rd, 2020-03-14:
@smitty97  you could try and manually modify the file. The change isn't very large.

marian42, 2020-03-14:
Sorry for not responding to this earlier, I was on vacation...

Thank you to @Zinip and @kylix-rd for reporting the issue and for providing a fix!

I just pushed an update that checks all `T...` commands against Octoprint's valid heater regex and discards the command if that check fails. You don't need to apply the patch anymore, you can just update the plugin through Octoprint.

Zinip, 2020-03-14:
@smitty97 
@kylix-rd 
Thats what I had done. First I compared the changes to the whole init.py here online just so I knew what I was looking at, then SSH'ed in to Octoprint, followed by:
~ cd oprint/lib/python2.7/site-packages/octoprint_preheat
~ cp __init__.py old_init.py  ;; just to have a copy of the old init.py just in case :)
~ sudo nano __init__.py
And from there once in the text editor I found the first line that needed to be removed, removed it, and added the lines that needed to be added, making sure to keep the spacing correct, and then did the same for the 2nd block of text. Ctrl+X'ed out, saving the changes, rebooted, and boom my button worked again!

@marian42 
It's np! Everyone has things they gotta do, besides that, the last thing I'm going to do is complain about someone that made something awesome for free went on vacation! I'm getting ready to try the update right meow and I'll tell you how it went. (Not doubting it at all, just thought you might like some confirmation) :p

commit df85f566 (2020-03-14): Check T commands for validity when parsing gcode

fixes #34