marian42/octoprint-preheat #36: Feature Request: Temperature offset

ubershy, 2020-03-30:
Thank you for this plugin. I'm using it since it appeared in Octoprint plugin list. I would like to propose a new feature.

## Temperature offsets

### Description
Allows the user to set negative offsets for bed and nozzle temperatures separately in the plugin's settings.
For example, when nozzle temperature offset is is set to **30** degrees, and gcode initial temperature **250** degrees for nozzle, preheat button will preheat nozzle only to **220** degrees. **(250 - 30 = 220)**.

### Why is it needed

1. To prevent filament oozing from nozzle, but still keep temperature close enough to target temperature.
2. Usually the first layer is printed with higher nozzle temperature than the rest of the print for better bed adhesion. Preheating nozzle with negative offset will solve these associated problems:
   1. Degradation of plastic inside nozzle which may occur if the plastic is kept at higher than normal temperature for long enough time. May cause clogging of nozzle.
   2. Degradation of PTFE tube in some extruders. Personally I have Anet A6 printer with stock extruder. When printing ABS I print the first layer at 248 degrees. For me it significantly improves bed adhesion, but causes degradation of inner PTFE tube inside extruder. I always try to minimize time of nozzle having such high temperature, so I can replace PTFE tube less often. It's also safer for health as overheated PTFE tube releases very bad chemicals in air. And that's why I simply can't use preheat button when printing ABS.

### Suggestions for implementation
Add new section in in the plugin's settings. Let's call it **Negative temperature offsets**, for example.

Add two new numeric fields in this section:
**Nozzle temperature offset**
**Bed temperature offset**

Default value for each field: **0**
Deny input of negative values for safety reasons.

Maybe add description.

=)

kb-, 2020-05-04:
A better way to implement this would be to offer an option for simultaneous preheating finish time, by starting the preheating of each element at the right time.
1- Learn the heating curve of the different elements at first use (T vs time); sequential the first time; repeat learning if a new file has higher target temperatures; offer an option to reset learning
2- Calculate the optimal start delay of each element to reach the target temperature simultaneously
3- Start heating each element at the calculated time

marian42, 2020-05-11:
Going through these issues I just saw that I never replied to this. Thank you @ubershy for this well written and detailed request. I haven't forgotten about it. I'm currently short on time, but I'll implement it once I find the time. If someone else wants to work on this, I'll happily accept a pull request.

Andrei-Pozolotin, 2020-06-01:
alternative approach would be to expose configurable % temperature multiplier:
```
Nozzle Factor: [80]%
Hotbed Factor: [95]%
```

Maroon-Cat, 2020-12-17:
Let me to bring this plugin to life. I'll write it soon.

Maroon-Cat, 2021-02-12:
It was the first time ever I tried to use GitHub, thus I had plenty of tryings.

marian42, 2021-03-16:
Implemented in 9c7644284a760c8db81f699aed4257eff25e3f27

closed 2021-03-16 by marian42

PowerWiesel, 2021-03-16:
Thank you very much for this new feature. I love it :smile:

ubershy, 2021-03-17:
Thank you kindly, @Maroon-Cat  ♥