marian42/octoprint-preheat #39: Only heats Bed.

gary325, 2020-08-02:
From log:
2020-08-02 13:01:12,144 - octoprint.plugins.preheat - INFO - Preheating bed to 60.0.
2020-08-02 13:02:53,257 - octoprint.plugins.preheat - INFO - Preheating complete.

The extruder is never "turned" on.

0.5.1

marian42, 2020-08-02:
Did you uncheck tool preheating in the plugin settings? If not, please share an example gcode file.

gary325, 2020-08-03:
Bed and Extruder checked, Enclosure unchecked

[image]

Beginning GCODE:

;FLAVOR:Marlin
;TIME:3376
;Filament used: 1.03436m
;Layer height: 0.1
;MINX:79.001
;MINY:89.661
;MINZ:0.1
;MAXX:141
;MAXY:127.648
;MAXZ:13
;Generated with Cura_SteamEngine 4.6.1
M82 ;absolute extrusion mode
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

;*** Start Preheating ***
M140 S60 ; start preheating the bed
M104 S200 ﻿T0 ; start preheating hotend
G28 ; home
M190 S60 ; heat to Cura Bed setting
M109 S200 ﻿T0 ; heat to Cura Hotend
;*** End Preheating ***
​

After hitting the Preheat button (with above GCODE loaded, from terminal:
Recv: ok
[...]
Send: M117 0% L=-/129
Recv: ok
[...]
Send: M117 Preheating...
Recv: ok
Send: M140 S60.0
Recv: ok
[...]

Bed got to temp, extruder still at room temp;

Hit the Cool button:
From terminal:
Send: M104 S0
Recv: ok
Send: M140 S0
Recv: ok

marian42, 2020-08-05:
I just tested your gcode and it works as expected, heating the bed and tool.
I don't know why it works on my Octoprint, but not on yours. Maybe it's something in the rest of the gcode file or another plugin. But both seem unlikely. What happens if you save the gcode snippet you posted as a gcode file and try the preheat button with that?

Stady234, 2020-08-06:
Maybe I am doing something wrong... but I can not get your posted "preheat" section to appear in ANY gcode files... even after all my addons 'process' the files...  Do you have that section in your slicer?  Are you having it make variables to make the adjustments?  Have you tried removing that section and see if preheat works normal?

What are the rest of your preheat setting look like? i do have fallback temps in half of my setups.. but it never affected any.. here are the rest of my settings.. 
[image]

marian42, 2020-08-06:
Looks like there was a misunderstanding. I was suggesting that you simply take the gcode you posted and save it in a text file with a .gcode extension. That way, you'd be testing it the same way I did. That gcode file wouldn't print anything, but it could help with debugging.

Stady234, 2020-08-06:
ok, well.. with that... I copied the selection of gcode... web synced to each of my 5 octopi... each time saving a different temp to make sure my fallback temp was not the supplied values... and in each case, it preheated both hotend and bed to the specified file values...

Not sure if/how that will help the OP, but I am willing to help test anything else...  
--
These are fresh updated to the newest Octopi 1.4.2 that was released...

DamaxusAwe, 2020-08-09:
I have the same issue as the OP, RPi 3B+ on 1.4.2.  Printer is an Ender 3 Pro, using the BTT SKR Mini E3 v1.2 mainboard on Marlin 2.0.5.3. When the current print job is finished, I will disable all my plugins and add them back one at a time and see if that is the issue.  Otherwise maybe there is something conflicting in my Marlin build.

DropTarget, 2021-02-02:
I have the same issue.  Was this ever resolved?

DropTarget, 2021-02-02:
Found my problem. Hex editor revealed garbage characters in M104 line added by Cura. Re-edited Cura machine settings and extra characters were eliminated. Works great now.

closed 2021-02-15 by marian42