marian42/octoprint-preheat #41: Custom GCode on timeout

towlerg, 2020-12-12:
I assume "After Preheating is complete" means nozzle and bed up to temp. Would it be possible to add custom GCode on timeout.

I find that when I've almost finish in CAD, I go to OP and hit preheat. But then as always I underestimate how long it'll take before slicer ready and Preheat times out. I understand that I can change the timeout but if I can find a period that suits all cases.

Thanks in advance,   George

marian42, 2021-01-03:
I don't know how the timeout feature works in Octoprint, but it's not a part of this plugin, so I think this plugin isn't the right place to change that.

Maybe a plugin like this does what you want: https://plugins.octoprint.org/plugins/heatertimeout/

towlerg, 2021-01-04:
Sorry, I have confused your plugin with BettererHeaterTimeout. Damn, too many preheat plugins. Do I close this or do you?

closed 2021-02-15 by marian42