marian42/octoprint-preheat #43: Receive popup that preheat cancelled, but it is actually continuing.

evannadeau, 2020-12-19:
Hello, I appear to have spotted a bug.  I am on Octoprint 1.5.2, and have several times spotted a popup stating preheat cancelled due to being manually changed, yet it continues. I had (now disabled for testing) the sequential heating enabled, and this message I believe appears right when the nozzle preheat kicks in.  I'll post if I find any more info.

[image]

evannadeau, 2020-12-19:
Update: I just started another print, and I received the message immediately.  This is with the sequential heatup disabled, I'll try my next print with it on.  My suspicion is this message comes up when the nozzle heat starts.

marian42, 2021-01-03:
This message should appear if you change the temperature menually while it preheats. I assume you didn't do that. If the message appears immediately, that's a bug. Does it happen all the time or only sometimes? Do you have any options in the plugin that aren't the default? Do you have any other Octoprint plugins installed?

closed 2021-02-15 by marian42