marian42/octoprint-preheat #49: [FR] Option to use M190 and M109

lightmaster, 2021-03-07:
I'd like an option to use M190/M109 instead of M140/M104 so that the Preheat button will activate PRINTER_EVENT_LEDS in Marlin.

marian42, 2021-03-16:
Can you achieve the desired result if you add a M150 command in the "Before preheat" gcode area in the Preheat plugin configuration?

lightmaster, 2021-03-17:
Not really, since the PRINTER_EVENT_LEDS uses a changing gradient of colors as its shifts from blue to purple to red basically like a colored progress bar and it can also be set to sequentially change each pixel one by one as the temp increases.

I was thinking that a simple checkbox in the settings to switch between using M140/M104 as default, or use M190/M109 if you want it to activate PRINTER_EVENT_LEDS code. Wouldn't change anything for users who don't want this, and gives the option to use it for users that would be interested.

marian42, 2021-03-19:
I just pushed an update that adds this feature. Please let me know how it works.

lightmaster, 2021-03-19:
It'll be Monday before I'm home again, but I'll let you know when I can.

lightmaster, 2021-03-24:
Heating up works good.

When using this option, in order to allow cooldown, the cooldown button needs to send the `M108` command and then send the cooldown commands.

marian42, 2021-03-25:
I added the M108 command for the cooldown button. It seems to work fine on the virtual printer, but I can't test it with the real printer. Please report back if it works as expected

closed 2021-03-25 by marian42

lightmaster, 2021-03-25:
It works great to cancel the preheat. LEDs do not reset to default though, so gonna open a new Feature Request to be able to set custom GCode to run after the cooldown code.

commit 93bc1991 (2021-03-25): Add a gcode command for the cooldown sequence if the M109 option is enabled

fixes #49