marian42/octoprint-preheat #52: “Cool” button shows snowflake icon, but still says “Preheat”

npburns224, 2021-04-12:
When the extruder and/or bed are already hot the button shows a snowflake indicating the option to turn the heaters off, but it’s still labeled “Preheat”

[image]

marian42, 2021-04-13:
On the screenshot it looks like you're using a dark theme. Can you disable the dark theme and confirm that the bug still occurs using the default theme?

LazeMSS, 2021-04-14:
@marian42 I think its related to my plugin UI Customizer theme:
https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/136

To make the buttons more responsive i add a wrapper around all button labels/text that hides the text and only show the icons on small screens.

This causes you text node to break:
https://github.com/marian42/octoprint-preheat/blob/93bc1991aa6273531b17e126c4100a3bb112323d/octoprint_preheat/static/js/preheat.js#L132

I will add a PR

marian42, 2021-05-31:
Fixed in 6142458c44c95bd6bcbca988330e294115215cb9

closed 2021-05-31 by marian42