marian42/octoprint-preheat #55: Preheat using wildly high temperatures for the heated tool

Deses, 2021-05-17:
[image]

I want to report an issue that I've seen every so often that could be potentially dangerous.
Sometimes the plugin sets the target temp to 395ºC and shortly after to 260ºC and it stays there unless the user chooses to press "Cool". You can see the spikes in the attached graph.

I was recording a video to show you proof but after disconnecting and reconnecting to the printer to clear the loaded gcode, the plugin started working fine again setting the correct temperature (200ºC). 

I've seen this behavior several times now and the gcodes are correctly set to 200ºC...

If it happens again I'll post a follow up video.

Deses, 2021-05-17:
**Versions used:**
OctoPrint 1.6.1
Python 3.7.3
OctoPi 0.17.0

Preheat Button (0.7.2)

**Printer:**
Ender 3 with stock hotend.

marian42, 2021-05-31:
This does look concerning! If it does work sometimes and doesn't work sometimes it's really difficult to diagnose. Did it happen again since you posted this report? Does it happen with all files or only with certain files?

Deses, 2021-05-31:
I haven't been printing much lately, I'm waiting for parts and I have the printer half disassembled, but tomorrow I'll run some tests if I can find a pattern. As a fellow programmer I understand how frustrating intermittent errors are. 

I really want to help you identify the issue, if there's really one! 

Does this plugin outputs any log? I can't check myself right now.

Deses, 2021-06-01:
I received the parts today, I've rebuild the machine and so far I haven't seen any issues so far. I've updated the plugin to version 0.8.0, too.

I also saw that the octoprint.log show the preheat logs... Unfortunately I no longer have the logs from 16 days ago. :(

I'll let you know when I see this behavior again! I'll try to record it then.

koningcool, 2021-07-07:
I have the same problem, I'm also on version 0.8.0 and it keeps happening.

Deses, 2021-07-07:
> I have the same problem, I'm also on version 0.8.0 and it keeps happening.

I'm "glad" you also had this issue, since that indicates that's not only a thing that happens to me alone.

I haven't printed in more than a month so that's why I haven't posted any follow up to this issue, but I hope I can provide feedback when I get back into printing.

Deses, 2021-08-07:
[image]

It's doing it again!

```
...
2021-08-07 02:02:10,983 - octoprint.util.comm - INFO - Firmware states that it supports emergency GCODEs to be sent without waiting for an acknowledgement first
2021-08-07 02:02:15,559 - octoprint.plugins.tracking - INFO - Sent tracking event printer_connected, payload: {'firmware_name': 'Marlin 2.0.7.2 (Nov 10 2020 04:02:24)', 'printer_port': None, 'printer_baudrate': 115200}
2021-08-07 02:02:21,816 - octoprint.plugins.preheat - WARNING - Error parsing heat command: M190 S[bed_temperature] 
2021-08-07 02:02:21,817 - octoprint.plugins.preheat - WARNING - Error parsing heat command: M109 S[temperature_0]   
2021-08-07 02:02:21,822 - octoprint.plugins.preheat - INFO - Preheating bed to 60.0.
2021-08-07 02:02:21,823 - octoprint.plugins.preheat - INFO - Preheating tool0 to 395.0.
2021-08-07 02:02:41,060 - octoprint.plugins.action_command_notification - INFO - Got a notification: Ender 3 Ready.
2021-08-07 02:02:42,871 - octoprint.plugins.preheat - WARNING - Error parsing heat command: M190 S[bed_temperature] 
2021-08-07 02:02:42,871 - octoprint.plugins.preheat - WARNING - Error parsing heat command: M109 S[temperature_0]   
2021-08-07 02:02:42,883 - octoprint.plugins.preheat - INFO - Preheating bed to 60.0.
2021-08-07 02:02:42,885 - octoprint.plugins.preheat - INFO - Preheating tool0 to 395.0.
2021-08-07 02:02:54,600 - octoprint.plugins.action_command_notification - INFO - Got a notification: Ender 3 Ready.
2021-08-07 02:04:11,329 - octoprint.plugins.preheat - WARNING - Error parsing heat command: M190 S[bed_temperature] 
2021-08-07 02:04:11,330 - octoprint.plugins.preheat - WARNING - Error parsing heat command: M109 S[temperature_0]   
2021-08-07 02:04:11,340 - octoprint.plugins.preheat - INFO - Preheating bed to 60.0.
2021-08-07 02:04:11,343 - octoprint.plugins.preheat - INFO - Preheating tool0 to 395.0.
2021-08-07 02:04:16,336 - octoprint.plugins.action_command_notification - INFO - Got a notification: Ender 3 Ready.
```

Looks like that when it cannot parse a heat command it goes crazy. 395 ºC holy sheet.