marian42/octoprint-preheat #59: Doesn't work with TouchUI

kyngs, 2021-12-29:
Even though it is shown working in README, it doesn't work. The button does not appear. Someone already posted this issue in TouchUI's repo, but the dev said it needs to be fixed here. [issue](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/400)
[image]
If you need some logs I will send them. Both plugins are the latest version.

kyngs, 2022-01-15:
It looks like this is caused by some other plugin, I did a fresh install only with touchui and preheat and it seems to appear now, I will try to find out which plugin causes this.

kyngs, 2022-01-15:
Ok this was caused by the dashboard plugin, you may close the issue or attempt to find the cause, I will leave it open.

JDCodes, 2022-10-10:
had the same issue with the dashboard plugin at my pi display. i fixed it with the Tab Order Plugin and moved the dashboard tab behind the temperature, control... and now i have the button back on my display.