marian42/octoprint-preheat #61: Preheating failed. Could not find any preheat commands in the gcode file. You can configure fallback temperatures for this case.

sourekv, 2023-08-28:
Hello,
for my file

[bolt_0.2mm_PETG_MK3S_1h49m.gcode.zip](https://github.com/marian42/octoprint-preheat/files/12451711/bolt_0.2mm_PETG_MK3S_1h49m.gcode.zip)

I have encountered following error:

```
Preheating failed.
Could not find any preheat commands in the gcode file. You can configure fallback temperatures for this case.
```

Could you please investigate the issue?

If there is something I can do. Let me know.
Thanks a lot.

Vlada

[octoprint-systeminfo-20230828095106.zip](https://github.com/marian42/octoprint-preheat/files/12451733/octoprint-systeminfo-20230828095106.zip)

KnobFaces, 2023-10-26:
In that GCode the heatup commands appear around line 1120 due to an embedded thumbnail. The default read is 1000 lines.

sourekv, 2023-10-27:
Thanks for explanation. I guess that fix is a feature request. Should I close this issue?

KnobFaces, 2023-10-27:
You can specify the number of lines to process in the options.

sourekv, 2023-10-27:
Great, Thanks

closed 2023-10-27 by sourekv